nesl/radar-camera-fusion-depth #1: Some errors when running the code

fanrz, 2023-05-28:
hi, thanks for your nice work!
when I ran the code, some errors came out. some module or import errors.
If you can improve the code, I will appreciate it!

thecyclone, 2023-05-28:
Hi @fanrz,

would you mind sharing the terminal output when you run the code? Thank you

thecyclone, 2023-07-06:
closed due to inactivity.

closed 2023-07-06 by thecyclone

preetispillai, 2023-08-16:
Can u plz help me out to resolve this error

Traceback (most recent call last):
  File "setup/gen_panoptic_seg.py", line 39, in <module>
    from detectron2 import model_zoo
  File "/home/lql2kor/cam_rad_fusion/lib/python3.8/site-packages/detectron2/model_zoo/__init__.py", line 8, in <module>
    from .model_zoo import get, get_config_file, get_checkpoint_url, get_config
  File "/home/lql2kor/cam_rad_fusion/lib/python3.8/site-packages/detectron2/model_zoo/model_zoo.py", line 9, in <module>
    from detectron2.modeling import build_model
  File "/home/lql2kor/cam_rad_fusion/lib/python3.8/site-packages/detectron2/modeling/__init__.py", line 2, in <module>
    from detectron2.layers import ShapeSpec
  File "/home/lql2kor/cam_rad_fusion/lib/python3.8/site-packages/detectron2/layers/__init__.py", line 3, in <module>
    from .deform_conv import DeformConv, ModulatedDeformConv
  File "/home/lql2kor/cam_rad_fusion/lib/python3.8/site-packages/detectron2/layers/deform_conv.py", line 11, in <module>
    from detectron2 import _C
ImportError: libtorch_cuda_cu.so: cannot open shared object file: No such file or directory