nesl/radar-camera-fusion-depth #5: Reproducing problem

ch5374, 2023-09-04:
Hi,

I trained the model according to your instructions, and the final result is as follows. (Max depth = 50m)

[image]

How can I achieve the performance in your paper?

Could you guys please provide the training log?

Thank you!

thecyclone, 2023-09-04:
Hi @ch5374,

Thank you for your interest in our work. The results seem very off from our model and there is not enough information in your post to guess how you can improve them. If you can do the following / provide the following information, I should be better able to debug your issue:

1. Have you tested our provided pre-trained models on your dataset? If you do not get same results as us, it means that there are some issues with your dataset.
2. Please provide the training settings for nuScenes (i.e. the bash file)
3. Please provide the name of the folder created under trained_fusionnet directory when your train your fusionnet model.

For our model, here are the trainings results at 2steps ( I am trying to see if I can find the full training log):
[Note that during training, we report results at a Max depth of 100m]
`135000  3036.157  6361.501    11.905    19.828`
`815000  2316.768  5295.748     8.912    15.726` -> Best model

ch5374, 2023-09-04:
Thanks for the quick reply!

Repond to your question:

   1. Yes! I tested your pre-trained models on dataset. 
       This is test result (run_fusionnet_nuscenes_test.sh) using 
       provided pre-trained weight in stage 2. 
 
[image]

       Looking at the results, it appears that there is no problem, at 
       least in the test dataset.

    2. Default setting you provided. The only difference is that 4 
        GPUs were used (export CUDA_VISIBLE_DEVICES="4,5,6,7").

    3. You mean the checkpoint_dirpath? 
 
        fus18project6ms1bn_16x448x448_lr0-1e3_100_aug0- 
        100_100_bri080-120_con080-120_sat080- 
        120_hflip_l1_sm000_wd000_outrm7- 
        150_dilate0_min1max100_lidar_loss200_interp_with_reproj
         
        Since I haven't touched anything, I don't think there will be any 
        problems.

My new question:
   
    1. Is it possible to reproduce performance in paper if I continue 
        learning with the default values ​​of your code in stage 2? 
        Because my model's learning progressed only to 343,000 
        steps, but your pre-trained model's learning progressed to 
        815,000 steps. There seems to be something wrong with this.

    2. Are the numbers of train, val, and test datasets 27430, 5869, 
        and 5858 respectively?

    4. If you find full training log, let me know. I want to know if my 
        343,000 steps results are reasonable.

Thank you in advance!

ch5374, 2023-09-04:
1. Thanks for the quick reply!

First of all, I want to let you know that I am only focusing on stage 2 right now.

Repond to your question:
1. asd 
2. sad

ch5374, 2023-09-04:
Thanks for the quick reply!
First of all, I want to let you know that I am only focusing on stage 2 right now.

Repond to your question:

1.Yes! I tested your pre-trained models on dataset. This is test result (run_fusionnet_nuscenes_test.sh) using provided pre-trained weight in stage 2. 
[image]

Looking at the test results, there appears to be no problem, at least in the test dataset.
2. Default setting you provided. The only difference is that 4 GPUs were used (export CUDA_VISIBLE_DEVICES="4,5,6,7").
3. You mean the checkpoint_dirpath? 

fus18project6ms1bn_16x448x448_lr0-1e3_100_aug0- 
100_100_bri080-120_con080-120_sat080- 
120_hflip_l1_sm000_wd000_outrm7- 
150_dilate0_min1max100_lidar_loss200_interp_with_reproj

Since I haven't touched anything, I don't think there will be any 
problems.

And this is my new question:
1. Is it possible to reproduce performance in paper if I continue learning with the default values ​​of your code in stage 2? Because my model's learning progressed only to 343,000 steps, but your pre-trained model's learning progressed to 815,000 steps. There seems to be something wrong with this.
2.  Are the numbers of train, val, and test datasets 27430, 5869, and 5858 respectively?
3. If you find full training log, let me know. I want to know if my 343,000 steps results are reasonable.

Thank you in advance!

thecyclone, 2023-09-05:
1. You can change the total number of epoch by tweaking the `learning_schedule` parameter in the bash file from 200 to 450.
2. Yes
3. You can see where our training was at 135,000 steps in my message above. If you are somewhere near, you should be on the right track.

ch5374, 2023-09-05:
Thank you!

I will try to continue learning by increasing the learning schedule.

closed 2023-09-05 by ch5374