nesl/radar-camera-fusion-depth #6: question about radar point cloud

harborsarah, 2023-09-07:
Dear author,

I am going through the code and would like to ask a question. In this function, merge_radar_point_clouds(), when you generate the accumulated depth map, you use the frames rather than accumulating through sweeps, right? Is there any specific reason of doing this? Since nuScenes also provide function like: 
```
pcs, times = RadarPointCloud.from_file_multisweep(self.nusc, sample, sensor_channel, \
                sensor_channel, nsweeps=required_sweep_count, min_distance=0.0, merge=False)
```
To accumulate through sweeps.

Thanks a lot for your answer.

thecyclone, 2023-09-07:
Hi @harborsarah ,

No specific reason, just wanted to keep the processing uniform between lidar and radar. You are free to accumulate however you like.

harborsarah, 2023-09-07:
> Hi @harborsarah ,
> 
> No specific reason, just wanted to keep the processing uniform between lidar and radar. You are free to accumulate however you like.

Okay I see. Thank you for your reply :)

closed 2023-09-07 by harborsarah