nesl/radar-camera-fusion-depth #7: question about 'setup/gen_panoptic_seg.py'

harborsarah, 2023-09-08:
Hi,

I have another question regarding the segmentation mask. When I run 'python setup/gen_panoptic_seg.py', the panoptic masks are only generated for the last frame in each scene. Masks for other frames are stored in data/semantic_segmentation folder.

Is there any reason for doing this? Or, actually, they should all be stored in 'panoptic_segmentation_masks' folder?

Since I also try to run bash setup_dataset_nuscenes.sh, and I get errors like:  

FileNotFoundError: [Errno 2] No such file or directory: 'radar-camera-fusion-depth/data/nuscenes_derived/panoptic_segmentation_masks/bd8edc7862984fd597f5931ca0ee20c0.npy'

Thanks for the help

thecyclone, 2023-09-08:
Hi @harborsarah ,

I am not sure I follow your question. If you look:
1. [We save the panoptic masks for all frames](https://github.com/nesl/radar-camera-fusion-depth/blob/75628de39b780065ec14c3f4e1a96da2ab318847/setup/gen_panoptic_seg.py#L130)
2. [We do this one more time for the last frame for the scene](https://github.com/nesl/radar-camera-fusion-depth/blob/75628de39b780065ec14c3f4e1a96da2ab318847/setup/gen_panoptic_seg.py#L148)

So we should have the panoptic masks for all frames.

harborsarah, 2023-09-08:
> Hi @harborsarah ,
>
> I am not sure I follow your question. If you look:
>
> 1. [We save the panoptic masks for all [...]

Yes, [but here, seg_path is 'data/nuscenes_derived/semantic_segmentation'](https://github.com/nesl/radar-camera-fusion-depth/blob/75628de39b780065ec14c3f4e1a96da2ab318847/setup/gen_panoptic_seg.py#L129C5-L129C5)

only for the last frame, [seg_path is 'data/nuscenes_derived/panoptic_segmentation_masks'](https://github.com/nesl/radar-camera-fusion-depth/blob/75628de39b780065ec14c3f4e1a96da2ab318847/setup/gen_panoptic_seg.py#L147)

thecyclone, 2023-09-08:
Thank you for pointing that out! Seems to have been a discrepancy.

I have fixed it in the lates commit. Can you pull it on your end and try it out now?

harborsarah, 2023-09-08:
> Thank you for pointing that out! Seems to have been a discrepancy.
> 
> I have fixed it in the latest commit. Can you pull it on your end and try it out now?

Okay, I see. That makes sense. I will try now. I think this time will succeed! Thanks a lot for your quick reply! :)

closed 2023-10-02 by harborsarah

xiaobh1519, 2024-01-10:
I met this problem, too, so have you solve it? If so, how did you solve?

thecyclone, 2024-01-13:
More info here: https://github.com/nesl/radar-camera-fusion-depth/issues/19