nesl/radar-camera-fusion-depth #10: How to use nuscenes v1.0-test for model testing?

zhuzhaohuan, 2023-10-12:
Dear author, I downloaded v1.0 test, pretrained models 'model-195000' and 'model-815000'. 
Am I running the 'run_fusionnet_nuscenes_test.sh' file directly?  Thank you for your reply！

thecyclone, 2023-10-12:
1. bash setup_dataset_nuscenes_test.sh -> to create the nuscenes derived dataset
2. bash setup_dataset_nuscenes_radarnet_test.sh -> using the pretrained RadarNet model
3. bash run_fusionnet_nuscenes_test.sh -> using the pretrained FusionNet model

zhuzhaohuan, 2023-10-12:
Hi. When I was running the second step, I found that calling forward() lacked a 'bounding_boxes_list' parameter, which caused an error during runtime.
After comparing the 'setup_dataset_nuscenes_radarnet.py' file, I found that there is no statement to obtain 'bounding_boxes_list' in this file. May I ask how to solve this problem?
The specific code is as follows:
[image]
[image]

zhuzhaohuan, 2023-10-13:
After I rerun, the following error occurred and the program is stuck here. It seems that the denominator has a situation of 0,  I don't know how to modify it.
[image]

thecyclone, 2023-10-13:
Can you please paste the entire error message?

zhuzhaohuan, 2023-10-13:
This is all the information output at runtime.
[image]

thecyclone, 2023-10-13:
Unfortunately, I am will not be able to debug using this message. However, I can point you to some general directions that you need to look into:
1. Check if the data is generated correctly in the previous steps like ground truth etc. Ground truth should not be empty.
2. Check if the output depth being generated is not empty: https://github.com/nesl/radar-camera-fusion-depth/blob/8cc44527fa1764d9dd7fa7e3ca74d81a66ab9b6e/setup/setup_dataset_nuscenes_radarnet_test.py#L302
3. Check if the inputs to the model are empty or not.
4. Finally, if you are not able to debug it but still want to proceed, you can remove the `--run_evaluation` from the bash file -> However, this is not recommended.

zhuzhaohuan, 2023-10-13:
1. Thank you for your guidance. I will try to solve this problem.
2. emm...This parameter was not set too much, so I ran it using a py file. Do these two methods have any impact?

thecyclone, 2023-10-13:
Yes, I would recommend to go back and run all the bash scripts one by one (from the first one mentioned in this thread). If you do not follow all of our instructions, there may be some discrepancies.

closed 2023-10-30 by thecyclone