nesl/radar-camera-fusion-depth #11: FileNotFoundError when running  gen_panoptic_seg.py

zhuzhaohuan, 2023-10-16:
Hi, when I run the gen_panoptic_seg.py file,  the FileNotFound error occurred. 
I'm not sure if it's because I'm running under Windows, so I added '../' before the path code, But there was still an error.
[image]

[image]

zhuzhaohuan, 2023-10-16:
The problem has been resolved, and  you need to create your own 'nuscenes_derived/panoptic_segmentation_maskds' folder

thecyclone, 2023-10-16:
Hi @zhuzhaohuan,

I am bit curious here. Did you not generate the panoptic masks before proceeding to generating training dataset using radarnet before?

zhuzhaohuan, 2023-10-17:
Yes，I use command to generate the panoptic masks from the beginning. Besides, after I run the setup_dataset_nuscenes.sh,I found that the images in the nuscenes_derived/lidar folder are all black, is there a problem?

thecyclone, 2023-10-18:
No, they should not be empty. They are sparse so they may look all black on your screen. You should be able to load them using numpy and then check if they contain some values.

closed 2023-10-30 by thecyclone