nesl/radar-camera-fusion-depth #13: Some error about use multiple GPU to train

zhuzhaohuan, 2023-10-28:
Dear author,
There is no problem when I use a single GPU for training，but when I use 4 GPU and run 'train radarnet.sh', the following error occurred, I couldn't find any problems in the code, I hope you can provide suggestions. Thank you for your reply.
[image]
[image]

zhuzhaohuan, 2023-10-30:
When using multiple GPUs, if batch size=1, the following issues will be prompted. If the batch size > 1, the problem shown in the above figure will occur.
[image]

thecyclone, 2023-10-30:
Hi @zhuzhaohuan ,

Unfortunately, I do not have the means to recreate and debug your issue. I would suggest you try to debug it yourself or use a single GPU for training in the meantime.

closed 2023-12-06 by zhuzhaohuan