nesl/radar-camera-fusion-depth #14: Code error

yuzzcyy1, 2023-10-30:
Dear author，
When I run “run_radarnet_nuscenes.sh", the following error occurs:
  File "src/run_radarnet.py", line 66, in <module>
    run(restore_path=args.restore_path,
  File "/data1/rcf/radar-camera-fusion-depth-main/src/radarnet_main.py", line 974, in run
    output_depth, output_response = forward(
TypeError: forward() missing 1 required positional argument: 'bounding_boxes_list'
I will appreciate it if you can solve it!

SuGoe-Young, 2023-10-31:
I've seen the same problem in the past

you need to try git pull for new version

thecyclone, 2023-10-31:
Hi @yuzzcyy1,

Please pull the latest commit. I have tried to fix the issue for you. Since, I no longer have access to a GPU machine, I was not able to test it out. Please try it and let me know if it works for you. Thank you!

yuzzcyy1, 2023-10-31:
This method works, I've solved the problem, thank you  for your help

> Collaborator

closed 2023-10-31 by thecyclone