nesl/radar-camera-fusion-depth #17: Why inference result loss somthing data

YoungHsu1002, 2023-11-10:
I put nuscenes data which has 34179 for trainval, but after inference model my data is less then 34179?

thecyclone, 2023-11-10:
I am not sure what the reason for this might be, but you can check out the train-val split [here](https://github.com/nesl/radar-camera-fusion-depth/tree/main/data_split). If I remember correctly, it had 850 scenes and most scenes had 39/40 samples. Some scenes had even less samples.

YoungHsu1002, 2023-12-27:
Thx for your reply.I want to know about the dataset split which use for your study is official split or custom split?

thecyclone, 2024-01-02:
For the training and validation sets, as far as I can remember, we used the [splits provided here.](https://github.com/nutonomy/nuscenes-devkit/blob/master/python-sdk/nuscenes/utils/splits.py)

[Note that in nuScenes, the scene_id and the number in scene_name are different]

Our .pkl files [provided here](https://github.com/nesl/radar-camera-fusion-depth/tree/main/data_split) basically map the scene_names to the corresponding scene_ids.

The test set is the standard / official nuScenes test set: `v1.0-test`

Feel free to verify it for yourself and let me know if you find any discrepancies between the two.

YoungHsu1002, 2024-01-10:
I found that the original program logic did not take into account the last detection problem. The following is my modification of the original code:

```
def process_scene(args):
    '''
    Processes one scene from first sample to last sample

    Arg(s):
        args : tuple(Object, Object, str, int, str, str, int, int, str, bool)
            nusc : NuScenes Object
                nuScenes object instance
            nusc_explorer : NuScenesExplorer Object
                nuScenes explorer object instance
            tag : str
                train, val
            scene_id : int
                identifier for one scene
            lidar_camera_correspondence_dict : dict
                contains correspondence between a lidar token and its closest camera token
            panoptic_seg_dir : str
                directory where all the panoptic segmentation masks are stored
            first_sample_token : str
                token to identify first sample in the scene for fetching
            last_sample_token : str
                token to identify last sample in the scene for fetching
            n_forward : int
                number of forward (future) frames to reproject
            n_backward : int
                number of backward (previous) frames to reproject
            output_dirpath : str
                root of output directory
            paths_only : bool
                if set, then only produce paths
    Returns:
        list[str] : paths to camera image
        list[str] : paths to lidar depth map
        list[str] : paths to radar depth map
        list[str] : paths to ground truth (merged lidar) depth map
        list[str] : paths to ground truth (merged lidar) interpolated depth map
    '''

    tag, \
        scene_id, \
        lidar_camera_correspondence_dict, \
        panoptic_seg_dir, \
        first_sample_token, \
        last_sample_token, \
        n_forward, \
        n_backward, \
        output_dirpath, \
        paths_only = args

    # Instantiate the first sample id
    sample_id = 0
    sample_token = first_sample_token

    camera_image_paths = []
    lidar_paths = []
    radar_points_paths = []
    radar_points_reprojected_paths = []
    ground_truth_paths = []
    ground_truth_interp_paths = []

    print('Processing scene_id={}'.format(scene_id))

    # Iterate through all samples up to the last sample
    while sample_token != last_sample_token:

        # Fetch a single sample
        current_sample = nusc.get('sample', sample_token)
[... 334 lines cut ...]
        radar_points_reprojected = np.stack([
            points_radar_reprojected[0, :],
            points_radar_reprojected[1, :],
            depth_radar_reprojected],
            axis=-1)

        radar_points = np.stack([
            points_radar[0, :],
            points_radar[1, :],
            depth_radar],
            axis=-1)

        np.save(radar_points_reprojected_path, radar_points_reprojected)
        np.save(radar_points_path, radar_points)

    '''
    Move to next sample in scene
    '''
    sample_id = sample_id + 1
    #last sample end
    #=====================================================================================================
    print('Finished {} samples in scene_id={}'.format(sample_id, scene_id))
```

Through this change, I successfully obtained the complete val set information !

thecyclone, 2024-01-10:
Thank you, can you create a pull request with this so that I can review it?

thecyclone, 2024-03-02:
Closing due to inactivity.

closed 2024-03-02 by thecyclone