nesl/radar-camera-fusion-depth #19: Questions about Not such File or Folder when I run bash setup_dataset_nuscenes.sh

xiaobh1519, 2024-01-10:
Dear author, I met the problem below when I run "bash setup_dataset_nuscenes.sh", could you please help me deal with it? (I have finished running "python setup/gen_panoptic_seg.py" and generate "data/nuscenes_derived" folder)

Best wishes!

_Processing scene_id=744
Processing scene_id=756
Processing scene_id=750
Processing scene_id=762
Processing scene_id=768
Processing scene_id=774
Processing scene_id=780
Processing scene_id=786
Processing scene_id=792
Processing scene_id=798
Processing scene_id=804
Processing scene_id=810
Processing scene_id=816
Processing scene_id=822
Processing scene_id=828
Processing scene_id=834
Processing scene_id=840
Processing scene_id=846
multiprocessing.pool.RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/home/xbh/miniconda3/envs/nesl/lib/python3.8/multiprocessing/pool.py", line 125, in worker
    result = (True, func(*args, **kwds))
  File "/home/xbh/miniconda3/envs/nesl/lib/python3.8/multiprocessing/pool.py", line 48, in mapstar
    return list(map(*args))
  File "setup/setup_dataset_nuscenes_with_denseGT.py", line 1031, in process_scene
    points_lidar, depth_lidar = merge_lidar_point_clouds(
  File "setup/setup_dataset_nuscenes_with_denseGT.py", line 300, in merge_lidar_point_clouds
    main_camera_panoptic_mask = np.load(main_camera_panoptic_mask_path)
  File "/home/xbh/miniconda3/envs/nesl/lib/python3.8/site-packages/numpy/lib/npyio.py", line 407, in load
    fid = stack.enter_context(open(os_fspath(file), "rb"))
FileNotFoundError: [Errno 2] No such file or directory: '/home/akash/Documents/Radar-camera-depth-completion_mask/data/nuscenes_derived/panoptic_segmentation_masks/932f6a27f9574a76911bee95e0a9d29f.npy'
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "setup/setup_dataset_nuscenes_with_denseGT.py", line 1200, in <module>
    pool_results = pool.map(process_scene, pool_inputs)
  File "/home/xbh/miniconda3/envs/nesl/lib/python3.8/multiprocessing/pool.py", line 364, in map
    return self._map_async(func, iterable, mapstar, chunksize).get()
  File "/home/xbh/miniconda3/envs/nesl/lib/python3.8/multiprocessing/pool.py", line 771, in get
    raise self._value
FileNotFoundError: [Errno 2] No such file or directory: '/home/akash/Documents/Radar-camera-depth-completion_mask/data/nuscenes_derived/panoptic_segmentation_masks/932f6a27f9574a76911bee95e0a9d29f.npy'

thecyclone, 2024-01-11:
Hi,

have you tried this: https://github.com/nesl/radar-camera-fusion-depth/issues/11

xiaobh1519, 2024-01-11:
Thanks for quickly reply, It's a pity that i have tried this : https://github.com/nesl/radar-camera-fusion-depth/issues/11
but, my bug occurs when i run"bash setup_dataset_nuscenes.sh" , and that means I have runned "gen_panoptic_seg.py" successfully, so could you please help me with this error ?
Furthermore, could you please show me your folder structure within the radar-camera-fusion-depth-main/data folder?

[image]

thecyclone, 2024-01-11:
It seems like you are not modifying the location of your panoptic_segmentation folder in the bash file: https://github.com/nesl/radar-camera-fusion-depth/blob/bb7a2291a214598a7826910cc3cdc02e08891cda/bash/setup_dataset_nuscenes.sh#L10

You can see that in your error it mentions: `FileNotFoundError: [Errno 2] No such file or directory: '/home/akash/Documents/Radar-camera-depth-completion_mask/data/nuscenes_derived/panoptic_segmentation_masks/932f6a27f9574a76911bee95e0a9d29f.npy'` <- this is my directory.

xiaobh1519, 2024-01-14:
> It seems like you are not modifying the location of your panoptic_segmentation folder in the bash file:
> 
> https://github.com/nesl/radar-camera-fusion-depth/blob/bb7a2291a214598a7826910cc3cdc02e08891cda/bash/setup_dataset_nuscenes.sh#L10
> 
> You can see that in your error it mentions: `FileNotFoundError: [Errno 2] No such file or directory: '/home/akash/Documents/Radar-camera-depth-completion_mask/data/nuscenes_derived/panoptic_segmentation_masks/932f6a27f9574a76911bee95e0a9d29f.npy'` <- this is a directory on my computer and not yours.
> 
> I have also updated the file to avoid any confusion. You can pull the latest commit and try again.

OK, Thanks for your reply and I think there may be some error with my data so I will try to pull the latest commit and update my data now and give the feedbacks after I try.

xiaobh1519, 2024-01-14:
> It seems like you are not modifying the location of your panoptic_segmentation folder in the bash file: [...]

I'm so sorry, It still dosen't work after I pull the last commit, the new errorr is "OpenBLAS blas_thread_init: pthread_create failed for thread 63 of 64: Resour..." could you please come up with another method or help me via remote screen like using Todesk? I am feeling extremely anxious, and I really want to replicate your project. I am truly grateful for your assistance.

thecyclone, 2024-01-14:
> > It seems like you are not modifying the location of your panoptic_segmentation folder in the bash file:
> > https://github.com/nesl/radar-camera-fusion-depth/blob/bb7a2291a214598a7826910cc3cdc02e08891cda/bash/setup_dataset_nuscenes.sh#L10
> > 
> > You can see that in your error it mentions: `FileNotFoundError: [Errno 2] No such file or directory: '/home/akash/Documents/Radar-camera-depth-completion_mask/data/nuscenes_derived/panoptic_segmentation_masks/932f6a27f9574a76911bee95e0a9d29f.npy'` <- this is a directory on my computer and not yours.
> > I have also updated the file to avoid any confusion. You can pull the latest commit and try again.
> 
> I'm so sorry, It still dosen't work after I pull the last commit, the new errorr is "OpenBLAS blas_thread_init: pthread_create failed for thread 58 of 64: Resource temporarily unavailable OpenBLAS blas_thread_init: RLIMIT_NPROC 4096 current, 2061512 max OpenBLAS blas_thread_init: pthread_create failed for thread 59 of 64: Resource temporarily unavailable OpenBLAS blas_thread_init: RLIMIT_NPROC 4096 current, 2061512 max OpenBLAS blas_thread_init: pthread_create failed for thread 60 of 64: Resource temporarily unavailable OpenBLAS blas_thread_init: RLIMIT_NPROC 4096 current, 2061512 max OpenBLAS blas_thread_init: pthread_create failed for thread 61 of 64: Resource temporarily unavailable OpenBLAS blas_thread_init: RLIMIT_NPROC 4096 current, 2061512 max OpenBLAS blas_thread_init: pthread_create failed for thread 62 of 64: Resource temporarily unavailable OpenBLAS blas_thread_init: RLIMIT_NPROC 4096 current, 2061512 max OpenBLAS blas_thread_init: pthread_create failed for thread 63 of 64: Resource temporarily unavailable OpenBLAS blas_thread_init: RLIMIT_NPROC 4096 current, 2061512 max" could you please come up with another method or help me via remote screen like using Todesk? I am feeling extremely anxious, and I really want to replicate your project. I am truly grateful for your assistance.

Quick googling the error shows that `OpenBLAS is attempting to create multiple threads, but is being restricted by the system's limit on the number of processes a user can create (RLIMIT_NPROC)`. I think this is something you would need to figure out on your own. Maybe start by reducing the `--n_thread` parameter to `1`.

xiaobh1519, 2024-01-21:
> Quick googling the error shows that `OpenBLAS is attempting to create multiple threads, but is being restricted by the system's limit on the number of processes a user can create (RLIMIT_NPROC)`. I [...]

Thanks! I got it！

closed 2024-01-21 by xiaobh1519