nesl/radar-camera-fusion-depth #21: Why does the remaining training time appear negative after loading the pre trained model

Cajeput, 2024-02-27:
I downloaded the pre trained model from the paper and loaded it, but after training for a period of time, the remaining time became negative, and the current training step count exceeded the total training step count, so I did not stop training, but continued training. What should I pay attention to when loading pre trained models

thecyclone, 2024-02-27:
Hi @Cajeput ,

I would encourage you to look at the bash files more closely and see how their inputs are being ingested in the main code.

All our training bash files have a param called `learning_schedule` that determines for how many epochs the training will run for: https://github.com/nesl/radar-camera-fusion-depth/blob/1e2cd8f4af5909bf2242acc6fb8caf031ccb7363/bash/train_radarnet_nuscenes.sh#L31

If this param is set to `100`, it means that your training will run for 100 epochs. If you are loading a model that has already been trained for more than `100` epochs, the training time left will become negative and the training step count will exceed the total step count. You can change the logging logic to overcome this problem.

Cajeput, 2024-02-29:
Thank you for answering my question. May I ask how to draw the colored image of Figure 4 and the gradient legend next to it in the paper

thecyclone, 2024-02-29:
Hi, you can look in this folder: https://github.com/nesl/radar-camera-fusion-depth/tree/main/tools

You might need to make some changes depending upon what you want.

Cajeput, 2024-03-04:
Okay, I have another question: how to set the input data without RadarNet and how to train FusionNet directly and simply when comparing without RadarNet and with RadarNet.

Cajeput, 2024-03-06:
Hello, author, I have the following questions:
1. Is the pre-trained model you provided (i.e., the best model you provided) based on minimizing loss or minimizing MAE/RMSE?
2. Is the output_depth and response obtained before training FusionNet based on the RadarNet model with the minimum loss or the RadarNet model with the minimum MAE/RMSE?
Looking forward to your reply!

thecyclone, 2024-03-06:
> Hello, author, there are several questions:
> 
> 1. Is the pre-trained model you provided (i.e., the best model you provided) based on loss minimization or MAE/RMSE minimization?
> 2. Is the output_depth and response obtained before training FusionNet based on the RadarNet model with the minimum loss or the RadarNet model with the minimum MAE/RMSE?
> 3. Addition: It is mentioned in the article that RadarNet is based on minimizing binary cross entropy loss, and FusionNet is based on minimizing L2.  However, in the output log of the code, the selection of the best model is based on three of the four error metrics being better than the previous model (i.e., based on MAE/RMSE minimization).  Therefore, I am somewhat confused
>    Looking forward to your reply!

Hi @Cajeput,

Please make a separate issue per question.

Training Loss for Radarnet: https://github.com/nesl/radar-camera-fusion-depth/blob/1e2cd8f4af5909bf2242acc6fb8caf031ccb7363/src/radarnet_model.py#L153

Training Loss for Fusionnet: https://github.com/nesl/radar-camera-fusion-depth/blob/1e2cd8f4af5909bf2242acc6fb8caf031ccb7363/src/fusionnet_model.py#L172

Bash file that sets the fusionnet loss: https://github.com/nesl/radar-camera-fusion-depth/blob/1e2cd8f4af5909bf2242acc6fb8caf031ccb7363/bash/train_fusionnet_nuscenes.sh#L43 

we are using L1 loss for Fusionnet so not sure where you are seeing `L2` loss. 

The losses are the training objective and the selection of the best model is based on the performance on the validation set. You do not select the best model based on lowest training loss but based on the performance on the validation set. This is the reason why we have validation set in the first place.

I would highly encourage you to read the paper and the code thoroughly as most of your questions can be answered by doing that. 

As to training on Fusionnet directly without the Radarnet, you would need to update the input of the Fusionnet with the radar points directly.

Cajeput, 2024-03-06:
Thanks for your answer. I chose the optimal model based on the performance of the validation set(step 25000):
Step=340000/342900 Time Elapsed=36.29h  Time Remaining=0.31h
Loss=0.07223
Validation results:
    Step       MAE      RMSE      iMAE     iRMSE        # Output  # Intersection  # Ground truth
  340000  1626.429  2489.933    11.959    16.042       24686.832        1559.635      144752.595
Best results:
    Step       MAE      RMSE      iMAE     iRMSE        # Output  # Intersection  # Ground truth
   25000   860.035  1052.873    29.757    34.059        6543.495         377.288      144752.595
Validation results:
    Step       MAE      RMSE      iMAE     iRMSE        # Output  # Intersection  # Ground truth
  342900  2155.759  3477.064    11.876    16.291       39597.455        3262.629      144752.595
Best results:
    Step       MAE      RMSE      iMAE     iRMSE        # Output  # Intersection  # Ground truth
   25000   860.035  1052.873    29.757    34.059        6543.495         377.288      144752.595

But most of the responses and depths obtained according to the optimal RadarNet on the validation set do not contain valid information,(many of the generated responses and radar are blank)

I have seen the MAE/RMSE of the best model you provided mentioned in other issues. The MAE/RMSE of the RadarNet model I obtained on the verification set was all small, but the response and depth were poor; On the contrary, RadarNet with minimal loss can obtain better response and depth. Can you ask me what the reason is?

thecyclone, 2024-03-06:
> Thanks for your answer. I chose the optimal model based on the performance of the validation set(step 25000): Step=340000/342900 Time Elapsed=36.29h Time Remaining=0.31h Loss=0.07223 Validation [...]

There is a tradeoff between the number of points and the error. For example, in the case you have mentioned, at MAE 860, the # points is ~6500. At MAE 2100, the # points is ~39500. At MAE 1600, the # points is ~24500. You are free to pick whichever model you want. In our case, we prioritized a balance of # points and error.

I am not sure what you mean by `Can you ask me what the reason is?`

Cajeput, 2024-03-07:
Okay, I'll read the code a little bit more, thanks.

closed 2024-03-07 by Cajeput