nesl/radar-camera-fusion-depth #22: Some questions about using sparse depth maps directly as FusionNet inputs

Cajeput, 2024-03-07:
In the code you provide, depth and response are required for FusionNet training to get the input depth, but how is this response obtained when the sparse depth map is entered directly into FusionNet?
[input_depth = torch.cat([depth, response], dim=1)
 output_depth = fusionnet_model.forward(image=image, input_depth=input_depth)](url)

Cajeput, 2024-03-07:
At present, the sparse depth map obtained by directly projecting the original radar points onto the image plane has been obtained. How to input it to Fusionnet? (In the case of no RadarNet)

Cajeput, 2024-03-08:
Hello, can you help me answer the above question? @thecyclone

Cajeput, 2024-03-20:
I still hope you can answer my question:
My question is that you mentioned in your paper "simply training FusionNet using raw radar points", but the inputs to FusionNet include semi dense radar depth maps and corresponding confidence maps. Can you explain how to train FusionNet using raw radar points?

zhuzhaohuan, 2024-05-13:
Hello, I noticed in your previous issue that you downloaded and utilized a pre-trained model. I was wondering if you experienced any issues with the pre-trained model in the scripts 'run_fusionnet_nuscenes_test.sh' and 'setup_dataset_nuscenes_radarnet_test.sh', as depicted in the figure below?
[image]

zhuzhaohuan, 2024-05-13:
> I still hope you can answer my question: My question is that you mentioned in your paper "simply training FusionNet using raw radar points", but the inputs to FusionNet include semi dense radar [...]

Hello, I noticed in your previous issue that you downloaded and utilized a pre-trained model. I was wondering if you experienced any issues with the pre-trained model in the scripts 'run_fusionnet_nuscenes_test.sh' and 'setup_dataset_nuscenes_radarnet_test.sh', as depicted in the figure below?
[image]

thecyclone, 2024-05-17:
Unfortunately, I do not have access to a GPU machine so I cannot give you the code or test anything for you. You will need to  implement it yourself. Thanks

closed 2024-05-17 by thecyclone