nesl/radar-camera-fusion-depth #23: Some questions about supervision during training and validation

Cajeput, 2024-03-08:
Hello author, I have a few questions that I hope can be solved:
dgt mentioned in the paper refers to the lidar depth map provided by nuScenes, and dacc refers to the accumulated depth map. However, in the code, the lidar depth map provided by nuScenes is saved under `nuscenes_derived/lidar/`. However, `nuscenes_derived/ground_truth/` stores the accumulation depth map. 
Why is the ground truth in the paper and the ground truth in the code not the same thing? Why is the accumulation depth map not directly described as d_gt in the paper? What about describing the lidar depth map provided by nuScenes as d_lidar?
Looking forward to your reply. @thecyclone

thecyclone, 2024-03-10:
Hi @Cajeput,

If you look at the [testing script](https://github.com/nesl/radar-camera-fusion-depth/blob/1e2cd8f4af5909bf2242acc6fb8caf031ccb7363/bash/run_fusionnet_nuscenes_test.sh#L14), you will notice that the ground truth being used is indeed the ground truth provided by nuScenes. All the test results are reported using this.

So yes, `dgt` in the paper refers to lidar in the code.

We chose to call the accumulated depth as ground_truth since we were using it as the ground_truth during training.

Cajeput, 2024-03-11:
Hello author, I have another question: in your paper, you mentioned the construction of binary classification labels by dacc, and you also mentioned the construction of ygt by interpolation depth map, and the supervision of RadarNet by ygt. What are the purposes of these two labels? Why do you need binary classification tags built by dacc when you already have ygt monitoring?

thecyclone, 2024-03-11:
Page 5 of the paper explains why we are doing this. Please refer to it.

Cajeput, 2024-03-11:
Thank you. You mentioned that ygt is obtained by interpolating the depth map, but where RadarNet training parameters are set (Page 6:RadarNet (Stage-1)), you say that ygt is obtained by dacc, can you explain why?

thecyclone, 2024-03-11:
My first answer in this thread is about Fusionnet supervision and not Radarnet supervision. I encourage you to read sections 4 and 5 of the paper to clear any confusion that you might have.

Cajeput, 2024-03-12:
Hello, after carefully reading the code, I have found the following questions:
When training FusionNet, there are the following fragments in the `bash` script:
```
--train_ground_truth_path \
    training/nuscenes/nuscenes_train_ground_truth_interp.txt \
--train_lidar_map_path \
    training/nuscenes/nuscenes_train_ground_truth.txt \
```
The path of the interpolation depth map is saved in nuscenes_train_ground_truth_interp.txt, and nuscenes_train_ground_truth is the accumulated depth map of 161 frames scanned by LiDAR. However, in train_fusionnet.py, these two command line parameters are assigned values to "-- train_ground_truth-path" and "-- train_lidar_map_path" respectively:
```
parser.add_argument('--train_ground_truth_path',
    type=str, required=True, help='Path to file that contains training paths for ground truth')
parser.add_argument('--train_lidar_map_path',
    type=str, required=True, help='Path to file that contains training paths for single lidar scan maps')
```
My question is: why is the accumulated depth map passed as a command line number to a single LiDAR scan -- train_lidar_map_path?

thecyclone, 2024-03-12:
1. Please see the loss function of any network that you are concerned about in the paper -- Radarnet / Fusionnet
2. Look at the corresponding code -- Look at where we are calculation loss in the code
3. The corresponding values will help you understand what is what.

`ground_truth` and `ground_truth_interpolated` are basically two forms of dense supervisions available to us for training. The third type of supervision is called `lidar`. I encourage you to try training with a combination of any 2 of the 3. We chose the combination that delivered the best results for us empirically. 

Since it has been a while since we worked on this project, I am not certain what should be the correct combination in the bash file.

closed 2024-03-14 by Cajeput