nesl/radar-camera-fusion-depth #24: data_utils.py don't have function

CC-owner, 2024-03-12:
radar-camera-fusion-depth-main/setup/setup_dataset_nuscenes_radarnet.py

thecyclone, 2024-03-12:
Here's the function: https://github.com/nesl/radar-camera-fusion-depth/blob/1e2cd8f4af5909bf2242acc6fb8caf031ccb7363/setup/data_utils.py#L63

Not sure why you're getting the error. Are you using the bash (.sh) file to run it?

thecyclone, 2024-03-12:
Okay, Now I see the issue. You're not following the instructions.

To setup the dataset, you're supposed to run the bash file provided here: https://github.com/nesl/radar-camera-fusion-depth/blob/main/bash/setup_dataset_nuscenes.sh

and not the python file. Please follow the instructions.

Here's the load image function if you want it: https://github.com/nesl/radar-camera-fusion-depth/blob/1e2cd8f4af5909bf2242acc6fb8caf031ccb7363/src/data_utils.py#L167

CC-owner, 2024-03-13:
Oh! Thanks for replying.

closed 2024-03-13 by CC-owner