nesl/radar-camera-fusion-depth #25: Questions about the results of quantitative analysis

Cajeput, 2024-03-14:
Hello, may I ask how the error results (MAE,RMSE) of other baselines (DORN,RC-PDA, etc.) are obtained during quantitative analysis? Looking forward to your reply. @thecyclone

thecyclone, 2024-03-14:
Page number 6: section 6 of the paper.

thecyclone, 2024-03-19:
What do you mean by training completed? You should load the current best checkpoint and keep training till the loss stops decreasing over many steps. You can modify the learning_schedule to train for longer: https://github.com/nesl/radar-camera-fusion-depth/blob/1e2cd8f4af5909bf2242acc6fb8caf031ccb7363/bash/train_fusionnet_nuscenes.sh#L42

Also, your choice of RadarNet can affect the final result.

Just to be sure, have you tried to download the pre-trained weights and evaluate them? This will help us ensure that your dataset is configured correctly.

Cajeput, 2024-03-20:
Thank you. I trained and evaluated according to the parameters you provided in the code, and found that the optimal checkpoint hardly changed when training FusionNet to 130,000 steps (the optimal checkpoint number you provided was 815,000 steps). In addition, the final evaluation results at 50, 70 and 80m are different from your results in the paper (for example, your MAE at 50m is 1727.7, but the result I got is 2050.) I have the following questions:
1. What are the possible reasons for this result?
2. In my training process, the optimal checkpoint hardly changes. Can the result be optimized by extending the training time as you said?
Looking forward to your reply.

thecyclone, 2024-03-20:
**have you tried to download the pre-trained weights and evaluate them? This will help us ensure that your dataset is configured correctly.**

Cajeput, 2024-03-20:
Yes, I downloaded the pre trained model you provided, and its result is consistent with the description in the paper.

thecyclone, 2024-03-20:
Okay, so that means your dataset is configured correctly. A few things you should do:

1. 135,000 steps training for Fusionnet is too low. Train it for 200,000 steps, pick the best checkpoint then train for another 200,000 steps. Repeat this process 3-4 times and I think you should get good results at around 7/800,000 steps. Don't stop the training pre-maturely.

2. Your choice of RadarNet can affect the final results.

3. You can also try lowering the learning rate by 50% or use another ground-truth to finetune further.

closed 2024-03-29 by thecyclone