nesl/radar-camera-fusion-depth #29: Not obtaining useful results from FusionNET

IoannaB04, 2024-06-18:
Hello authors,

I am following your instructions to run RadarNet and FusionNet but the results I get are not useful as you can see in the following image

[image]

Do you have any idea why?

thecyclone, 2024-06-21:
This look like the output of RadarNet. I can see some data in the image on the bottom left and bottom right sides.
You can check that the output is not empty using numpy. What do you mean by the results not being useful?

IoannaB04, 2024-06-26:
Could you please tell me where the depth maps from FusionNET are stored?

thecyclone, 2024-06-27:
Where ever you want to save them: https://github.com/nesl/radar-camera-fusion-depth/blob/main/bash/run_fusionnet_nuscenes.sh

Please read the code files as you run them. They contain the location where the folders are created.

closed 2024-07-01 by IoannaB04