nesl/radar-camera-fusion-depth #31: The Evaluation Metrics of Radarnet and the Learning Rate Adjustment Strategy of Fusionnet

ZZX10082, 2024-10-03:
Hello, thank you very much for your excellent work. During the process of reproducing your paper, I trained for 200 epochs and obtained a radar net weight. Its RMSE, MAE, and other indicators are similar to the evaluation results of the weight file you provided, but the output and Intersection are relatively low. What role do these two indicators play in judging network performance, and are higher indicators better? How should I adjust my learning strategy
[image]
[image]
 if I want to obtain a result that approximates the weights you provided

ZZX10082, 2024-10-03:
For the training of fusionnet, in the code you provided, it is trained using learning_rates 1e-3 and learning_stchedule 450, while the learning strategy mentioned in the paper is: "with a learning schedule 1e-3 for 400 epochs, then reduced to 5e-4 for another 50 epochs, and finally reduced 1e-4 for further fine-tuning After 450 rounds of training without changing the learning rate (i.e. without making any changes to the code in GitHub), I obtained the best results at step=695000. The network performance obtained is significantly different from the reference indicators in the paper. Do I need to modify the parameters according to the paper to adjust the learning strategy? And what you mentioned in your paper, "Training takes ≈ 36 hours for 75 epochs on a NVIDIA RTX A5000 GPU." "Training takes ≈ 36 hours for 200 epochs on a NVIDIA RTX A5000," is this just a reference for training time, rather than referring to the fact that Radarnet and Fusionnet only need to be trained for 75 and 200 epochs respectively
[image]
[image]

ZZX10082, 2024-10-13:
I have reduced m
[image]
y learning rate and continued training, but still cannot achieve satisfactory results. Can you help me? @thecyclone

zhang-xu0124, 2024-11-04:
> 我降低了学习率并继续训练，但仍然无法取得满意的结果。你能帮助我吗？@thecyclone ![f94b3a9da0f30c7ae4d822955d518264](https://private-user-images.githubusercontent.com/161538770/376007305-5b07f08e-9e2e-4cfa-97de-aac7b470fc67.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA2OTM1NjUsIm5iZiI6MTczMDY5MzI2NSwicGF0aCI6Ii8xNjE1Mzg3NzAvMzc2MDA3MzA1LTViMDdmMDhlLTllMmUtNGNmYS05N2RlLWFhYzdiNDcwZmM2Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwNFQwNDA3NDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ODA2ZTIzNjQyOWVjNmEzOWUyZmUxOGJmYzI1NTIzOTI5YTRkNWQzNWFjMDU3MzU5Yjc4OTZiM2U5Mjg2NzcyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.CoExDB7_XS_jXYQYHZI44wLgySKipms5Mq_9VAA2-DY)

Hello, I have been reproducing this article recently, but I have encountered some questions. Can I communicate with you? How can I contact you？

ZZX10082, 2024-11-07:
23s101094@stu.hit.edu.cn @zhang-xu0124